nrwiersma/ESP8266Scheduler #3: Adding "run_groups" and "data_spaces"

MikeFair, 2016-11-13:
Hi!

I've been working with the ESP8266Scheduler a bit and while I haven't used the official Arduino, or any other "Scheduler", libraries for comparison; I really like it!

One thing about the "one-big-loop" model is you have lots of control over the ordering of if/when things happen and they can share in-common data easily.  Scheduler largely eliminates that in favor of coroutines wrapped in their own classes that all run mostly segregated and seemingly in parallel.

I've written some basic/initial code to add a "run_groups" concept to the Scheduler which adds back some degree of those coordination capabilities while keeping everything nice and separated. :)

In a Task code, either setup() or the constructor(), you'd add something like ```setRunGroupID(N);``` where currently N is a number 0..7 or 255 (the default).
(The number of groups available is defined by the size of a bitfield used internally; I used an uint8_t.)

By assigning a Task a run_group_id, that Task will be delayed until all Tasks with lower run_group_ids have "succeeded" for this cycle.  "Success" means the Task considers its work completed and will wait for the next "cycle" before resuming.  So it could mean the Task "timed out" on what it was doing and didn't want to hold up downstream RunGroups from processing any longer.

RunGroup 0xFF is the default RunGroup, and it's also the exception to the rule. Tasks with run_group_id 0xFF use the current behavior - other run_groups do not wait for them, and they do not wait for anything else (other than their own shouldRun()).  

Excluding that exception, groups of Tasks will execute in the order of their run_group_id in a cycle.  Higher numbered RunGroups will delay until all Tasks in the lower RunGroups have marked themselves complete.  

When all the Tasks in all RunGroups have executed, the cycle_id number increments and the Task's completed state gets reset.

Here's some example from 3 Tasks to explain it more clearly.
2 Tasks "Acquire" data then 1 "Processes" it.

Notice the cycle_id (the first number) remains the same until all three Tasks have run and that the two "Acquiring" Tasks Finish before the "Processing" Task begins; and the cycle_id increments only after the "Processing" Task completes:
```
[Cycle_ID] ([Task_Id]) [Function]... [Internal Counter]
2	(1.1) Acquiring... 1
2	(1.2) Acquiring... 1
2	(1.2) Acquiring... 2
2	(1.1) Acquiring... 2
2	(1.2) Acquiring... 3
2	(1.2) Acquiring... 4
2	(1.1) Acquiring... 3
2	(1.2) Acquiring... 5
2	(1.2) Acquiring... 5 Done!
2	(1.1) Acquiring... 4
2	(1.1) Acquiring... 5
2	(1.1) Acquiring... 5 Done!
2		(5.1) Processing... 1
2		(5.1) Processing... 2
2		(5.1) Processing... 3
2		(5.1) Processing... 4
2		(5.1) Processing... 5
2		(5.1) Processing... 5 Done!
3	(1.1) Acquiring... 1
3	(1.2) Acquiring... 1
3	(1.2) Acquiring... 2
3	(1.2) Acquiring... 3
3	(1.1) Acquiring... 2
3	(1.2) Acquiring... 4
3	(1.1) Acquiring... 3
3	(1.2) Acquiring... 5
3	(1.2) Acquiring... 5 Done!
3	(1.1) Acquiring... 4
3	(1.1) Acquiring... 5
3	(1.1) Acquiring... 5 Done!
3		(5.1) Processing... 1
3		(5.1) Processing... 2
3		(5.1) Processing... 3
3		(5.1) Processing... 4
3		(5.1) Processing... 5
3		(5.1) Processing... 5 Done!
4	(1.1) Acquiring... 1
4	(1.2) Acquiring... 1
4	(1.2) Acquiring... 2
4	(1.2) Acquiring... 3
4	(1.1) Acquiring... 2
4	(1.2) Acquiring... 4
4	(1.2) Acquiring... 5
4	(1.2) Acquiring... 5 Done!
4	(1.1) Acquiring... 3
```

In this example, no Task had run_group_id 0xFF because it's output more cluttered and confused the example more so than helped it.

If there's interest, I can explain the internals of how it works a bit more; and discuss modifications for merging it.  While it's not "trivial", it's not a big diff.

One of the things that I put a lot of time into is keeping the amount of state required to as little as I could imagine (but no less :) ) and making the update functions as efficient/quick as I could (for instance, the new code never makes a run through the entire list of Tasks; it does it all "inline" as the Scheduler reaches them).

The SchedulerClass added 4 bytes; and the Tasks 2 bytes and 2 bools.

The only I couldn't do easily was get loop() to return a bool rather void.
I think that'd make the whole concept much easier/cleaner to work with.

Thanks!

nrwiersma, 2016-11-13:
Wow,

That is some awesome work. This method didn't even occur to me. Thanks very much. I greatly appreciate the effort.

Could you explain more about wanting loop() to return a bool?

MikeFair, 2016-11-13:
Sure, it's how a Task signals "Done for this cycle".
The Tasks all have a bool variable called "loop_complete" which is their signal to the Scheduler to stop calling them until the next cycle.  If the loop() function returned a bool, that return item would get placed in that loop_complete spot by the Scheduler.

This way, to the programmer, loop() returns "false" until that Task "succeeds" and returns "true".
They use the return code to indicate to the Scheduler the success/failure of the loop().

This way, if at the top of the loop() code it detects some failure condition like "External Device is not responding"; they can set a couple flags so they know to deal with that next time loop() is called and do a short circuit "return true;" instead of the double command "setLoopComplete(true); return;"

You can check out the diff here:
https://github.com/MikeFair/ESP8266Scheduler/commit/0bb21d625d0d3360227f60ae6c16012b03203dc2

The white space got a bit screwed up in some places, and a comment needs to be fixed. (Tasks indicate they are no longer active when loop_complete == true; the comment says false.)

And I'm pretty sure the two dedicated additional boolean variables added the Task can be eliminated.
By changing cycle_id and run_group_id on the Tasks to a struct with a 1-bit flag, and a 7-bit number; in place of an uint8_t, loop_complete can use the high bit on cycle_id, and active_run_group can use the high bit on run_group_id. 
Which eliminates yet another small piece of overhead on each Task. :)

nrwiersma, 2016-11-13:
Ok,

I am starting to get my head around it. Surely ```loop_complete``` could be made private and ```loop()``` could return a bool. You could then set ```loop_complete``` in ```loopWrapper()```?

Like:
```
void loopWrapper() {
    if (!setup_done) {
        setup();
        setup_done = true;
    }

    while(1) {
        this->loop_complete = loop();
        yield();
    }
}
```

Would that not solve that issue?

nrwiersma, 2016-11-13:
Perhaps some of the reset logic for the ```current_cycle_id``` and ```loop_complete``` could also be internalized into the task. Perhaps being a ```reset()``` and ```update()``` function. This may take some of the burden away from the Scheduler.

MikeFair, 2016-11-13:
```
void loopWrapper() {
    if (!setup_done) {
        setup();
        setup_done = true;
    }

    while(1) {
        this->loop_complete = loop();
        yield();
    }
}
```

Exactly!
And perhaps the group update functions should go there too.  
Currently they execute anytime the Task runs delay or yield too; which creates a side effect when loop_complete = true happens before a delay(); the loop() would "pause" and then pick up again just after the delay() on the next cycle.  Which makes perfect sense for what/how it's working; just not what I originally intended.

----

> Perhaps some of the reset logic for the current_cycle_id and loop_complete could also be internalized into the task. Perhaps being a reset() and update() function. This may take some of the burden away from the Scheduler.

I'm not opposed to the idea but I'm not sure it would work. 
I in fact actually started out by putting it in the Tasks following the Delay pattern; the problem was Task.h cannot see Scheduler.h to call any functions the SchedulerClass exported (like getCurrentCycleID()) and I didn't want to mess with that; I also saw that SchedulerClass was already a friend to Task so could mess with the internals of Task if it needed to; but not the other way around.  It also turned out the Task update code reduced to a couple if tests and a couple assignments in a function that already had to look at the internal state information of the Task anyway; so updating that information "while it was there" was an extremely "light" addition...

I thought briefly about separating the state and logic out into a new RunGroups.h that they both included; again that seemed too heavy handed to keep the feature "light" on the resources (though I honestly don't think it would actually be any more "expensive" than the current method and it would make a good "extension pattern" that could be followed by others so would be worth doing)...

To be as open/forthcoming as I can see about it; there are two things a Task could manage itself:
```
  // (1)
  if (scheduler_cycle_id != current->current_cycle_id) {
    if (task_run_group_id != 0xFF) current->loop_complete = false;
    current->current_cycle_id = scheduler_cycle_id;
  }

  // and (2) psuedocode
  if (the_tasks_run_group_is_an_active_group) {
    current->run_group_active = true;
  } else {
    current->run_group_active = false;
  }
```

In other words, (1) if the Scheduler's Cycle has moved from what the Task remembers; that's the signal to reset the loop_complete flag; and (2) the Scheduler indicates to the Task whether or not it's run_group is currently active by setting the flag.  

The Task then tests those flags in shouldRun().

Everything else would almost certainly be calling a function the SchedulerClass exported because that's where the group state is (because Tasks can't see information about other Tasks).

And let's keep in mind too regarding the term "burden"; it's going to be the same CPU cycles spent to do the work whether it happens in the Scheduler or in the Tasks, regardless of where it happens, because the work must be done.  The key point more important than "who" does the work is maximizing time spent in "loop()" (by minimizing CPU cycles other places); a Task's "shouldRun()" is not "loop()" and so therefore is as equally costly to "loop()" as the Scheduler is; I'd even argue that a Task's "shouldRun()" is an externalized extension of the Scheduler. :)

nrwiersma, 2016-11-13:
> I thought briefly about separating the state and logic out into a new RunGroups.h that they both included;

I really like this idea. It is unlikely to be very much more expensive after it has been compiled, but the level of clarity and intention would likely be worth it. I will take another look through the code this week to see if there is any more insight there from my side.

> And let's keep in mind too regarding the term "burden"; 

My use of the word burden was not to imply CPU cycles, but rather role and responsibility, and in general, design. It would probably be better to have each class manage as much of it own internals as possible. While the line is a little blury here cause we are reaching in for ```next``` and ```context```, the more we can centralize the logic of a Task in a Task, I would argue, the better. This is however opinion :) I will however take your point on giving the task less responsibility some more thought. Perhaps there is some middle ground here, I am not sure yet.

MikeFair, 2016-11-14:
> It would probably be better to have each class manage as much of it own internals as possible. While the line is a little blury here cause we are reaching in for next and context, the more we can centralize the logic of a Task in a Task, I would argue, the better. 

Agreed, the question then is "Are RunGroups an internal state to a Task? or are RunGroups part of the runtime environment the Task lives in?"  I never saw RunGroups as part of the logic of a Task; but I also didn't see "Delay" as internal to a Task either.

Given the model you're alluding to, it's seems Tasks are responsible for coordinating themselves and choosing to run or not run as they deem fit and the Scheduler serves as a Task list container and execution pump.  

"Delay()" isn't something the Scheduler enforces on the Task's behalf; it's something the Task chooses to honor because "shouldRun()" gives the Task the chance to pump state.  If a Task didn't want to honor "Delay()" it doesn't have to.  For instance, if it wanted "Interruptable Delay" which would delay for a timeout period unless some signal happened that triggered it to resume earlier... [Which btw is another really important function: bool WaitOn(SOME_SIGNAL_IDENTIFIER, time_out);]

Ok, here's the model I'm seeing; extending a Task's runnable features means coding up a Header File to implement logic and store shared state, and extending a Task's shouldRun rather than mess with the TaskPump itself.  This means "delay" should also be moved into a new Delay.h which Task.h includes; to serve as an example for how to "think" this way.

These are features that a Task includes, like add-on modules, and uses "shouldRun()" to feed them data.

FWIW, and to that point, here's what I see:
>> I thought briefly about separating the state and logic out into a new RunGroups.h that they both included;
> I really like this idea. It is unlikely to be very much more expensive after it has been compiled, but the level of clarity and intention would likely be worth it. 

Implement RunGroups.h with the 4 variables that are currently in the Scheduler and move the two functions into RunGroups.h. 

RunGroups.h will need to know to increment the cycle_id.
This was originally done in the Scheduler when it saw Scheduler::main was about to get Scheduled. 
I think the best approach to mimic this is having Scheduler::main call updateCycleID() in RunGroups.h. This avoids RunGroups.h storing how many Tasks there are or storing pointers to compare (which would be duplicated by other modules; storing the same data redundantly in multiple places).

Tasks then call the RunGroups.h update functions passing in their own task_id(this), run_group_id, and loop_complete during shouldRun() (Or they pass in the struct with the first bit a boolean (loop_complete), and a 7-bit run_group_id).

The update function in RunGroups.h will return a struct with a Boolean "group_active" bit, and a Cycle_ID.  The logic in those functions should work for this almost unchanged; and may in fact get simpler...

nrwiersma, 2016-11-14:
> Given the model you're alluding to, it's seems Tasks are responsible for coordinating themselves and choosing to run or not run as they deem fit and the Scheduler serves as a Task list container and execution pump.

Interesting point you make. While I was interested in centralizing logic in the Task, I think perhaps your point holds more weight. We should probably try moving more logic out of the Task and make the Scheduler responsible for this. 

I think I will split out a new branch to work on this when I get some time. Please feel free to contribute.

MikeFair, 2016-11-14:
An example of how possibly to do Delay.h:
```
#ifndef DELAY_H
#define DELAY_H

	struct DelayInfo {
		uint32_t delay_start = 0;
		uint32_t delay_ms = 0;
	}
	
    void delay(DelayInfo* delay_info, uint32_t ms) {
        if (ms) {
            delay_info->delay_start = millis();
            delay_info->delay_ms = ms;
        }

        yield(); // Do not return to the calling Task; move to the next Task
    }
	
	inline bool isDelayed(DelayInfo* delay_info) {
		return (delay_info->delay_ms != 0);
	}
	
	inline void updateDelay(DelayInfo* delay_info) {
		if (delay_info->delay_ms == 0) return;  // Optimize for the non-delayed case
	
		// This comparison is "rollover safe"
        uint32_t now = millis();
        if ((now - delay_info->delay_start) >= delay_info->delay_ms)
			delay_info->delay_ms = 0;
	}

	/*
	class Task {
		// ...
		DelayInfo delay_info;
		// ...
	}
    virtual bool shouldRun() {
		// ...
		// Pump the DelayInfo
		updateDelayInfo(&delay_info);
		if (isDelayed(&delay_info)) return false;
		// ...
	}
	void loop() {
		// ...
		delay(&delay_info, 1000);
		// ...
	}
	*/

#endif
```

MikeFair, 2016-11-14:
Following the same pattern, here's a first stab at RunGroups.h

(Note: I didn't test to see if these even compile; they are more a demonstration of "how" they might be done so you can get a sense/feel for the pattern of extending Tasks this way).

The general approach is:

1. A "feature" creates a struct to put its per Task data in.
2. The Task declares one of these structs and passes a pointer to it when the "features" functions are called.  These functions are mostly inline so they'll compile as if they are directly part of the Task's code.
3. The "feature" can use this pointer as a Task ID (since the address of each struct will be different for each Task) if it thinks it needs one.
4. It is expected for a feature to modify Task.h as required to call functions at their required times. So adding a feature isn't totally done in a black box with just the header/cpp file.  However none of the feature's "logic" or "state updates" happen in Task.h; that's what the feature's functions are for.
5. This keeps the logic for separate features isolated, and keeps Task.h "clean" in the sense that it is easy for people to work in a way that doesn't "mess up" the logic of other features.

This struct/struct address approach could also be reversed, and classes with their member functions used instead; structs and functions are just the first way I saw on how to "extend" a Task with a new feature that carries almost no overhead and can be "inline" so the compiler will treat them as if they were directly embedded in the Tasks' code.
(Skip to the bottom to see how this gets used in Task.h)

```
#ifndef RUNGROUPS_H
#define RUNGROUPS_H

	struct RunGroupInfo {
		uint8_t run_group_id = 0;
		
		uint8_t task_active:1 = 0;
		uint8_t run_group_active:1 = 0;
		uint8_t cycle_id:6 = 0;
	}

	uint8_t scheduler_cycle_id;
	uint8_t count_active_group_tasks;
	uint8_t scheduler_run_group_id;
	uint8_t lowest_active_run_group_id;
	uint8_t next_run_group_id;

	inline void updateTask(RunGroupInfo* run_group_info) {
		// Executes before loop()
		uint8_t task_run_group_id = run_group_info->run_group_id;
		uint8_t task_cycle_id = run_group_info->cycle_id;

		// First, check the cycle_id
		if ( task_cycle_id != scheduler_cycle_id ) {
			// If the Task's cycle_id doesn't match; reset task_active and set the current cycle_id
			if (task_run_group_id != 0xFF) run_group_info->task_active = true;
			run_group_info->cycle_id = scheduler_cycle_id;
		}

		// Next, update the Task's run_group_active flag
		// run_group_active is always true when the run_group_id == 0xFF
		run_group_info->run_group_active = (task_run_group_id == 0xFF) || (task_run_group_id == scheduler_run_group_id);

		// RunGroupId tracking: Calculate the next run_group_id, and the lowest run_group_id to use when the cycle wraps.
		if (task_run_group_id < lowest_active_run_group_id) lowest_active_run_group_id = task_run_group_id;
		if (task_run_group_id > scheduler_run_group_id && task_run_group_id < next_run_group_id) next_run_group_id = task_run_group_id;
	}

	void updateRunGroups(RunGroupInfo* run_group_info) {
		// This is called after the Task's loop() has run; 
		// updateTask(&run_group_info) reset task_active and cycle_id if required before loop() ran;
		// and if task_run_group_id was part of a currently active run_group, loop() could have set run_group_info->task_active

		// If this Task is task_active and also part of the current run_group, increment count_active_group_tasks; this group_id isn't done yet
		if (run_group_info->task_active && run_group_info->run_group_id == scheduler_run_group_id) count_active_group_tasks++;
	}

	void updateRunGroupsCycleId() {
		// If no Tasks in this group were active (all task_active == false), move to the next group_id
		if (count_active_group_tasks == 0) {			
			if (next_run_group_id == 0xFF) { // If no next highest run_group_id exists; then all run_groups have completed
				scheduler_run_group_id = lowest_active_run_group_id;
				scheduler_cycle_id++;        // As all run_groups have completed; incrment the scheduler_cycle_id
			} else {
				scheduler_run_group_id = next_run_group_id;
			}
		}
		count_active_group_tasks = 0;  		// reset count of still active tasks in this run_group
		lowest_active_run_group_id = 0xFF;	// reset lowest_active_run_group_id
		next_run_group_id = 0xFF;	        // reset next_run_group_id
	}

	/*
	class Task {
		// ...
		
		RunGroupInfo run_group_info;
		
		// ...
	}
    virtual bool shouldRun() {
		// ...
		
		// Check the run_group_active and task_active Status
		return (run_group_info.run_group_active && run_group_info.task_active);
	}
	void loopWrapper() {
		
		// ...
		
		while(1) {
			
			updateTask(&run_group_info);
			run_group_info->task_active = loop();
			updateRunGroups(&run_group_info)
			
			yield();
		}
	}
	
	class Scheduler {
		Scheduler::main::loop() {
			updateRunGroupsCycleID();
		}
	}
	*/

#endif
```

Lastly, I changed a couple things around in the RunGroups logic code above:
- The "loop_complete" flag was renamed to "task_active" and the logic is reversed; if loop() returns true, it will remain "active" and get called again.  When it's done, loop() should return false (no longer active).

- This moves updateTask() and updateRunGroups() to loopWrapper() instead of the Scheduler; the reason is the only time this information changes is before and after the completion of loop(); the Scheduler and "shouldRun()" are called much more often, like the between times during a delay().

- I eliminated the bit field for groups and track the "next_group_id" differently. Ther are now 256 available run_groups.

- I expect Scheduler::main's "loop()" function to call the RunGroups.h updateRunGroupsCycleID(); this avoids having to compare pointers or count Tasks.  This uses the Scheduler's "main" loop() to mark the beginning/ending of a "pass" through the Tasks.  Other modules might want something similar.

MikeFair, 2016-11-15:
>> Given the model you're alluding to, it seems Tasks are responsible for coordinating themselves and choosing to run or not run as they deem fit and the Scheduler serves as a Task list container and execution pump.

> Interesting point you make. While I was interested in centralizing logic in the Task, I think perhaps your point holds more weight. We should probably try moving more logic out of the Task and make the Scheduler responsible for this.

Sounds like a plan, :)
Now that I've done the work I've done, I think I get how it all works and see how to rearrange things.
I expect to stick with the header files approach to implementing the Scheduler features.

I'll share my results as I get them. :)

PS I'm also very curious how the official Arduino Scheduler library works.  Somehow keeps the main loop() function of a project active while it does its Task thing...  Do you know?

nrwiersma, 2016-11-15:
I have made a start in the "run_group" branch. I will hopefully have some time to take a stab at splitting run groups out tonight. Let me know what you think. 

Figured I would do it piece at a time, lastly moving the most of ```shouldRun()``` and friends into the Scheduler. I would like the Task to have some say as to weather it should run though. I use this in cases where no config has yet been provided, and think it is useful.

I am looking forward to seeing your results.

I don't know how the Arduino Scheduler works, it is possible though that it is a full multitasker, rather than a co-operative multitasker like this one. It would explain how it keeps running.

MikeFair, 2016-11-15:
> I would like the Task to have some say as to weather it should run though. I use this in cases where no config has yet been provided, and think it is useful.

I'll keep you posted, I agree shouldRun() is useful; a task's shouldRun() appears to fit better in Loopwrapper() where it selects whether or not loop() for the Task gets called.
Where it is now, it selects whether or not the task's continuation resumes (I.e. the task gets scheduled).

MikeFair, 2016-11-15:
First pass at moving things into the Scheduler and separating out "new features" into separate include files.

Here's the diff:
https://github.com/MikeFair/ESP8266Scheduler/commit/b9ea97453cea557fa58810d6002b0109394dae16

It seems to work as advertised; here's the output from my last run using both the delay and RunGroup features (though there's no timing information directly displayed in the output; the inconsistency in task output ordering is showing that differing delays were happening).

Look it over, I can make a pull request if you like.

nrwiersma, 2016-11-15:
Thanks. I'll take a look this week.

Have a good one.

MikeFair, 2016-11-17:
Here's a heads on the biggest/main changes:

1. Everything was redone in one shot:
    - The Scheduler now hosts both delay and RunGroups; 
    - The Task loopWrapper calls shouldRun() before the start of loop() (not during loop()) 
    - A new "can_schedule(Task* t)" [which replaces what shouldRun() was doing]; 
    - A new "class SchedulerTask : public Task" to serve as the new "main" for doing "Scheduler Work"

2. Delay.h provides a DelayInfo struct (Every Task gets an instance of this struct) and there's a "Delay" class with static methods that operate on that struct.  The Tasks also have local member functions that hide away the struct and external class functions.  Calls to "delay(10);" work like they always have.

3. RunGroups.h/cpp provides a RunGroupsInfo struct (again one per Task) and a RunGroups class with static methods and variables that operate on the struct and keeps track of the RunGroups state.  All 256 group ids are now valid (I eliminated the internal bitField tracker), and the ScheulderClass::main Task is used to detect when a full pass through the list of Tasks has happened.  (You'll see the addition of a "SchedulerTask" class)

4.  There's a new SchedulerTask.h which implements the class of ::main's Task.  Right now it only calls the function to update the RunGroup's cycle_id.  But I expect it to be useful for other modules that require periodic housekeeping which can happen between passes through all the active Tasks.

So far, all the Tasks in my testing are working exactly as the design intended.  :)

nrwiersma, 2016-11-17:
Following of from your run groups sample, I have had a run at simplifying the process. I think it worked out, but I haven't had time to run and debug this yet. Let me know what you think though. 

[run_groups](https://github.com/nrwiersma/ESP8266Scheduler/commit/d6835453bac4c119ae2701933ce695340975f3e0)

Will jump back on this asap. A little heck here at the moment.

nrwiersma, 2016-11-21:
So I had another crack at this this weekend, and the more I looked at it, the more it seemed too complex. For me this generally points towards a needed simplification or rethink. 

This is what I have come up with. If we treat a ```RunGroup``` as a mini scheduler, and abstract and share the methods used to determine if a ```Task``` can be run, we will simplify the issue greatly. The ```RunGroup``` can then be run like a ```Task```. When it is done, the next ```RunGroup``` or ```Task``` can be run.

I think in the end this will hold 2 advantages:
* The ```RunGroup``` logic is entirely encapsulated. It seems now to be sitting between the ```Scheduler``` and the ```Task```, which is a bit messy.
* The ```Scheduler``` can be used without run groups with no penalty. With ```RunGroup``` logic all over, not using them still incurs the logic of using them.

There are still some details I am working out.

MikeFair, 2016-11-21:
If you are saying something like:
```
    Task t1;
    Task t2;
    RunGroup rg1;

    rg1.addTask(t1);
    scheduler.addRunGroup(rg1);
    scheduler.addTask(t2);

```

Where a RunGroup is derived from, that makes sense to me.

MikeFair, 2016-11-22:
> The Scheduler can be used without run groups with no penalty. With RunGroup logic all over, not using them still incurs the logic of using them.

To be clear, not using any rungroups is very inexpensive.  When a Task is in RunGroup 0xFF, which is the default, most, if not all the rungroup functions have immediate short circuit returns.  

So the "cost" of not using RunGroups is pretty close to 3 "Branch if not equal" instructions and a couple assignments and adds to handle the function calls.  

Honestly, I'd compare the current implementation of delay that always does the millis() math regardless of whether or not there's an active delay as more expensive than not using RunGroups and having them there (at least the way I coded it).

BTW, I think there's a bug in the code you were looking; there's a point I call "yield()" but it's the system yield() and not the Task's yield().  So keep an eye out; I'll likely have more code to post soon.

ghislainborremans, 2017-02-21:
Hello
this is very interesting.
has this been finalised? 

best regards

nrwiersma, 2017-02-21:
Unfortunately not yet. I need to do a fair bit of restructuring to have then sit lightly in the library. I havnt had much time lately, and am hoping to get back to it soon.

MikeFair, 2019-02-01:
I take it that the idea didn't make it? ;-)

nrwiersma, 2019-02-01:
I dont have a lot of time for this project any more. I am basically just doing maintenance at this point. Feel free to add though :)