nrwiersma/ESP8266Scheduler #5: task not working after several hours

zenz, 2017-04-24:
As title, I'm trying to use Scheduler to manage different task, it works very well at the beginning, but after several hours, some task die(but others are still alive). below is my test code. Please kindly help me to figure out the problem.

`// 需要用到的库
#include <Arduino.h>
#include <Scheduler.h> // 分时管理
#include <ClickEncoder.h> // 旋钮按键
#include <Adafruit_Sensor.h>
#include <DHT.h> // DHT22 传感器库
#include <brzo_i2c.h> // I2C
#include <SH1106Brzo.h> // OLED

#define ENCODER_PINA D4 // 对应GPIO2，印刷版上D4
#define ENCODER_PINB D7 // 对应GPIO13，印刷版上D7
#define ENCODER_BTN D6 // 对应GPIO12，印刷版上D6
#define ENCODER_STEPS_PER_NOTCH 4

#define DHT_PIN D5  // 对应GPIO14，印刷版上D5
#define DHT_TYPE 11 // DHT22温湿度传感器

#define OLED_ADDRESS 0x3C
#define OLED_SCL D1
#define OLED_SDA D2

#define GLYPH_WIDTH 32
#define GLYPH_HEIGHT 64
// 图形
static const char glyphs[4][GLYPH_WIDTH * GLYPH_HEIGHT] PROGMEM = {
        { // 图案0，关闭状态
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0xFC, 0x3F, 0x00,
                0x00, 0xFF, 0xFF, 0x00, 0x80, 0xFF, 0xFF, 0x01, 0xC0, 0xFF, 0xFF, 0x03,
                0xE0, 0x7F, 0xFE, 0x07, 0xF0, 0x7F, 0xFE, 0x0F, 0xF8, 0x7F, 0xFE, 0x0F,
                0xF8, 0x79, 0x9E, 0x1F, 0xFC, 0x78, 0x1E, 0x1F, 0x7C, 0x7E, 0x7E, 0x3E,
                0x7C, 0x7E, 0x7E, 0x3E, 0x7E, 0x7F, 0xFE, 0x3E, 0x3E, 0x7F, 0xFE, 0x7C,
                0x3E, 0x7F, 0xFE, 0x7C, 0x3E, 0x7F, 0xFE, 0x7C, 0x3E, 0x7F, 0xFE, 0x7C,
                0x3C, 0x7F, 0xFE, 0x3C, 0x7C, 0xFE, 0x7F, 0x3E, 0x7C, 0xFE, 0x7F, 0x3E,
                0xFC, 0xFC, 0x3F, 0x1F, 0xF8, 0xF9, 0x8F, 0x1F, 0xF0, 0xE3, 0xC7, 0x0F,
                0xF0, 0x07, 0xE0, 0x0F, 0xE0, 0x1F, 0xF8, 0x07, 0xC0, 0xFF, 0xFF, 0x03,
                0x80, 0xFF, 0xFF, 0x01, 0x00, 0xFE, 0x7F, 0x00, 0x00, 0xF8, 0x1F, 0x00,
                0x00, 0xC0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00
        },
        { // 图案1，加热状态
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x01, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0xC0, 0x00, 0x00,
                0x00, 0xE0, 0x00, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00,
                0x00, 0xF8, 0x00, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x00, 0xFC, 0x00, 0x00,
                0x00, 0xFC, 0x00, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x00, 0xFE, 0x00, 0x00,
                0x00, 0xFE, 0x00, 0x00, 0x00, 0xFE, 0x00, 0x08, 0x00, 0xFE, 0x01, 0x0C,
                0x00, 0xFE, 0x01, 0x0E, 0x00, 0xFE, 0x01, 0x07, 0x00, 0xFC, 0x03, 0x07,
                0x20, 0xFC, 0x87, 0x07, 0x60, 0xFC, 0x87, 0x07, 0xE0, 0xFC, 0xCF, 0x0F,
                0xE0, 0xF9, 0xDF, 0x0F, 0xE0, 0xF9, 0xFF, 0x0F, 0xE0, 0xF9, 0xFF, 0x0F,
                0xE0, 0xF9, 0xFF, 0x0F, 0xE0, 0xF9, 0xFF, 0x1F, 0xF0, 0xF9, 0xFF, 0x1F,
                0xF0, 0xFD, 0xFF, 0x1F, 0xF8, 0xFD, 0xFD, 0x3F, 0xFC, 0xFF, 0xF9, 0x3F,
                0xFC, 0xFF, 0xF9, 0x7F, 0xFE, 0xFF, 0xF9, 0x7F, 0xFE, 0xFF, 0xF8, 0x7F,
                0xFE, 0xF7, 0xF8, 0x7F, 0xFF, 0xF7, 0xF8, 0x7F, 0xFF, 0x67, 0xD8, 0x7F,
                0xFF, 0x67, 0xC8, 0x7F, 0xFF, 0x27, 0xC8, 0x7F, 0xFF, 0x03, 0xC8, 0x7F,
                0xFF, 0x03, 0xC8, 0x7F, 0xFE, 0x03, 0x88, 0x7F, 0xFE, 0x01, 0x80, 0x3F,
                0xFC, 0x01, 0x00, 0x3F, 0xFC, 0x01, 0x00, 0x1F, 0xF8, 0x01, 0x00, 0x0F,
                0xF0, 0x01, 0x00, 0x07, 0xE0, 0x03, 0x80, 0x03, 0x80, 0x07, 0xC0, 0x00,
[... 165 lines cut ...]
        else oled.displayOff();
}
} oled_task;

// 主设置程序
void setup() {
        Serial.begin(115200);

        Serial.println("");

        delay(1000);

        // 加入各个任务
        Scheduler.start(&encoder_task);
        Scheduler.start(&dht_task);
        Scheduler.start(&oled_task);

        // 启动任务的执行
        Scheduler.begin();
}

// 主循环
void loop() {
}
`

nrwiersma, 2017-04-24:
Hi,

My best guess is there may be an overflow in the delay time, and therefore ShouldRun returns false. Perhaps try delaying that task by 1 and see if that fixes stuff.

zenz, 2017-04-24:
@nrwiersma 1ms? 1s? this encoder_task is depend on it's speed, cannot delay too long.

nrwiersma, 2017-04-24:
1ms would be enough to test if this is the bug.

nrwiersma, 2017-04-24:
The other option is to just return true from ```shouldRun```. That would also test this.

zenz, 2017-04-24:
@nrwiersma Thank you for your help. I found the solution as you suggested.
modify a bit to make encoder_task to wait a little bit longer then the problem solved.

nrwiersma, 2017-04-24:
Awesome. I will close this issue then. If you have any other issues, let me know

zenz, 2017-04-25:
oops, my bad, the task still dies after over night. so still need to analyst the issue.

nrwiersma, 2017-04-25:
I would suggest setting ```shouldRun``` to return ```true``` and see where that takes you. If that is not the issue, I would start Serial logging the free memory. Perhaps it is running out. This is unfortunately a difficult problem to debug.

zenz, 2017-04-25:
@nrwiersma shouldRun already return true, but not working.

nrwiersma, 2017-04-25:
Ok. My guess is that it has nothing to do with ESP8266Scheduler and something to do with your loop code. ```shouldRun``` is the only thing in the scheduler that will stop a Task from running. I would suggest you begin debugging your loop function.

zenz, 2017-04-25:
below is the current encoder task, it's quite simple that I cannot figure out where's the problem.

`class EncoderTask : public Task {
protected:
void setup() {
        encoder.setDoubleClickEnabled(true);
        encoder.setButtonHeldEnabled(false);
        encoder.setButtonOnPinZeroEnabled(false);
}
void loop()  {
        ClickEncoder::Button btn;
        static uint32_t lastService = 0;
        
        if (lastService + 2500 < micros()) {
                lastService = micros();
                encoder.service();

                static int16_t encLast, encValue = 0;
                encValue += encoder.getValue();
                if(encValue != encLast) {
                        if(isDisplayOn) { // 只有在屏幕点亮的时候才调整温度，避免误操作
                                targetTempCH = (encValue > encLast) ? targetTempCH + 1 : targetTempCH - 1;
                                if(targetTempCH > 38) targetTempCH = 38;
                                if(targetTempCH < 10) targetTempCH = 10;
                        }
                        encLast = encValue;
                }

                btn = encoder.getButton();
                if (btn != ClickEncoder::Open) {
                        switch (btn) {
                        case ClickEncoder::Clicked:
                                isDisplayOn = !isDisplayOn;
                                break;
                        case ClickEncoder::DoubleClicked:
                                if(isDisplayOn) { // 只有在屏幕点亮的时候操作切换模式才有效，避免误操作
                                        currentMode += 1;
                                        if(currentMode > 3) currentMode = 0;
                                }
                                break;
                        }
                }
        }
}
bool shouldRun() {
        bool run = Task::shouldRun();
        return true;
}
} encoder_task;`

nrwiersma, 2017-04-25:
I would start by commenting out block 2 and 3, leaving ```encoder.service();```. If that does not break, uncomment then next block. It is a bit verbose, but is the easiest way to figure out where the problem exists.

closed 2017-07-30 by nrwiersma