nrwiersma/ESP8266Scheduler #8: Memory usage.

a-d-j-i, 2018-01-11:
Is possible that using:
Class Bla : Task {
} bla;
eats more ram (like 4k) than:
class Bla : Task {};
static Bla bla;

nrwiersma, 2018-01-11:
Is there any reason you think this? 

I would not imagine it would, the size of a class instantiation is the same, regardless of it is declared static or not.

a-d-j-i, 2018-01-14:
Just adding a task seems to take 4k ram space.

class MemPrinter: public Task {
public:
	virtual ~MemPrinter() {
	}
	virtual void loop() {
		Serial.println(ESP.getFreeHeap());
		delay(2000);
	}
};
MemPrinter mem_printer;
MemPrinter mem_printer2;
MemPrinter mem_printer3;
MemPrinter mem_printer4;
void setup(void) {
	Serial.begin(115200);
	delay(100);
	//Scheduler.start(&mem_printer);
	Serial.println("Ready");
	Serial.print(F("MEM END: "));
	Serial.println(ESP.getFreeHeap());
	//Scheduler.begin();
}
void loop(void) {
	Serial.println(ESP.getFreeHeap());
	delay(2000);
}

nrwiersma, 2018-01-14:
Wow, 4k seems heavy for a class. I don't have my rig setup at the moment, I will have to look into this sometime this week.

Thanks for reporting.

a-d-j-i, 2018-01-14:
Probably part of the problem came from Task.context that is a cont_t that have a 
stack[CONT_STACKSIZE / 4];
and
#ifndef CONT_STACKSIZE
#define CONT_STACKSIZE 4096
#endif
I'm going to investigate further...

nrwiersma, 2018-01-14:
Thanks for looking into it.

If this is the case, there is unfortunately nothing to do about it. The cont_t is a critical part in being able to save and restore task context.

nrwiersma, 2018-09-07:
I assume this is resolved. Please reopen if this is still an issue.

closed 2018-09-07 by nrwiersma

dbuezas, 2021-02-01:
This maybe explains why I get a OOM error in the console (activated debug for SSL+HTMLClient) when making "heavy" use of the library.
 
* Without using the library (obviously no multitasking) have `36440 bytes free` in the heap before connecting to the host
* With 5 tasks running, I measure `11008` and the connection cannot be made.

nrwiersma, 2021-02-02:
Yes, that seem about right.