nrwiersma/ESP8266Scheduler #9: Delay in subroutines

ACB1337, 2018-06-18:
Subroutines are not calling the correct delay function in Task but rather the default delay function that returns control to the esp.

nrwiersma, 2018-06-18:
Can you explain more about what your are seeing?

ACB1337, 2018-06-19:
```
void foo()
{
  delay(10 * 1000); // does not call Task::delay
}
class Foo : public Task {
 virtual void loop() final
 {
   foo();
 }
}s_fooTask;
```
In this code the daly in foo() will not call Task::delay which results in all other tasks being suspended for 10 seconds

nrwiersma, 2018-06-19:
It would not call `Task::delay` as it is not in the task. It is an external method, as therefore would use the external delay. It has no reference to the task.

ACB1337, 2018-06-19:
Yeah but thats kinda problematic if you can only call subroutines inside the Task the whole longjmp solution becomes kinda redundant

nrwiersma, 2018-06-19:
What your proposed solution here? 

I tend to disagree with your argument though. Expecting `delay` in an external function to suddenly call the `Task::delay` would be weird and wrong. The idea is for the each `Task` to contain its own logic, hence it has its own `setup` and `loop` functions. It is meant to be a contained unit.

An example of perhaps what you are trying to is as follows:
```
#ifndef LIGHT_H
#define LIGHT_H

#include <Arduino.h>
#include <Task.h>

#define LIGHT_STATE_ON 930
#define LIGHT_STATE_OFF 0

class Light : public Task {
public:
    Light(uint8_t pin) : pin(pin), goal(0), state(0), steps(0), stepDelay(0) {}

    void setup() {
        pinMode(pin, OUTPUT);
        analogWrite(pin, state);
    }

    void loop() {
        int16_t state = state + ((goal - state) / steps);
        analogWrite(pin, state);

        if (state == goal) {
            steps = 0;
            return;
        }

        steps--;

        if (steps > 0) {
            delay(stepDelay);
        }
    }

    bool shouldRun() {
        bool run = Task::shouldRun();

        return run && steps > 0;
    }

    void set(uint16_t goal) {
        this->goal = goal;
        this->steps = 0;
        this->stepDelay = 0;
    }

    void set(uint16_t goal, uint16_t steps, uint16_t delay) {
        this->goal = goal;
        this->steps = steps;
        this->stepDelay = delay;
    }

protected:
    uint8_t pin;
    uint16_t goal;
    uint16_t state;
    uint16_t steps;
    uint16_t stepDelay;
};

#endif
```

ACB1337, 2018-06-19:
No that does not help with my case. I have to call external code that calls delay and wanted to run other stuff during that time. Hence the whish for delay to work from any function. I guess I will have to write something myself or try to get [this](https://github.com/anmaped/esp8266-scheduler) scheduler running

nrwiersma, 2018-06-19:
Fair enough. Good luck.

closed 2018-06-19 by nrwiersma