nrwiersma/ESP8266Scheduler #11: Doesnt work if use function => while (Client.connected())

adamrabbani, 2018-07-25:
i use this function => while (Client.connected()) for client-server communication..
when esp run the function.. other function delayed until client connected..

nrwiersma, 2018-07-25:
Can you give a code example?

adamrabbani, 2018-07-26:
#include <TimeLib.h>
#include <Scheduler.h>

void Lampu_Kedip(){

if(Keadaan_Relay[0] == 2){   
    
    timer2.run();
    }

  if(Keadaan_Relay[1] == 3){
     timer3.disable(0);
    timer3.restartTimer(0);
    
    }else  if(Keadaan_Relay[1] == 2){
        
   timer3.enable(0);
   timer3.run();
    }    
    

  }

class Start: public Task {
protected:
    void loop()  {
  timer1.run();
   Lampu_Kedip();
    }
} print_task;

class Client: public Task {
public:
    void loop() {
timer0.run();
    }
} mem_task;

void setup()
{

    Scheduler.start(&Start);
    Scheduler.start(&Client);

......................some code declaration to connect wifi

   timer0.setInterval(2000L, ClientServer);
   timer1.setInterval(100L, Kontrol);

.........................................
   
   Scheduler.begin();
}

    void ClientServer()
  {

    if (Client.connect(Server, Port))
  {
  
  
String a;
    root.printTo(a);
          
         Client.println(a);
          Client.flush();
          
**while (Client.connected())**
    {
      if (Client.available())
      {
        (Client.readStringUntil('\n')).toCharArray(json, num1);
        DataServer();
      }
    }
    
    }
    else
  {
    
    Client.stop();
 
  }
  }

adamrabbani, 2018-07-26:
#include <TimeLib.h>
#include <Scheduler.h>

void Blink(){

if(Blink_0[0] == 2){   
    
    timer2.run();
    }

  if(Blink_0[1] == 3){
     timer3.disable(0);
    timer3.restartTimer(0);
    
    }else  if(Blink_0[1] == 2){
        
   timer3.enable(0);
   timer3.run();
    }    
    

  }

class Start: public Task {
protected:
    void loop()  {
  timer1.run();
   Blink();
    }
} print_task;

class Client: public Task {
public:
    void loop() {
timer0.run();
    }
} mem_task;

void setup()
{

    Scheduler.start(&Start);
    Scheduler.start(&Client);

......................some code declaration to connect wifi

   timer0.setInterval(2000L, ClientServer);
   timer1.setInterval(100L, Kontrol);

.........................................
   
   Scheduler.begin();
}

    void ClientServer()
  {

    if (Client.connect(Server, Port))
  {
  
  
String a;
    root.printTo(a);
          
         Client.println(a);
          Client.flush();
          
**while (Client.connected())**
    {
      if (Client.available())
      {
        (Client.readStringUntil('\n'));
      }
    }
    
    }
    else
  {
    
    Client.stop();
 
  }
  }

adamrabbani, 2018-07-26:
'Client Server' set to 2s and 'Kontrol' set to 100ms 
but when i run.. 
waiting 2s for 'Client Server' .. after that 'Kontrol'
so Kontrol function run (waiting for server + 100ms ),.. not every 100ms..

adamrabbani, 2018-07-26:
'Client Server' set to 2s and 'Kontrol' set to 100ms 
but when i run.. 
waiting 2s for 'Client Server' .. after that 'Kontrol'
so 'Kontrol' function run time is (waiting for server + 100ms ),.. not every 100ms..

adamrabbani, 2018-07-26:
Kontrol function just bellow:

void Kontrol(){
Serial.println("Test")
}

nrwiersma, 2018-07-26:
That is to be expected as this is a co-operative multitasker. Only one piece of code runs at a time, and you code must yield to another task or end the loop which is an implicant yield. I think you think this is a pre-emptive multitasker, which would likely do what you expect.

adamrabbani, 2018-07-26:
ohh.. based priority multitasking..
but how can i change the priority to run function ..
Thanks

adamrabbani, 2018-07-26:
Oh.. i see...
Thanks you.

closed 2018-07-26 by adamrabbani