nrwiersma/ESP8266Scheduler #13: Problems with: while(WiFiMulti.run() != WL_CONNECTED){}

GiampaMan, 2018-12-04:
Hi!
You have done a great job, this scheduler is really fine!
I´m a beginer.

I'm doing a code, but when arrive a part like this: 
`while(WiFiMulti.run()` != WL_CONNECTED) 
    {
    }
`
The part is ignored.

I did two tests:

- First, I put the while section on general void setup (before indicating the task to do and before begining scheduler).

- Second, puting this on a task to do
Two tests, same results.

I will appreciate all your help!

nrwiersma, 2018-12-05:
HI, thanks very much.

If you are getting the same results in the task and out, then i think we can rule the scheduler out.
From what I can tell on the code, it likely means your wifi is connected, which would immediately exit the while loop.

Does that answer your question?

GiampaMan, 2018-12-05:
> HI, thanks very much.
> 
> If you are getting the same results in the task and out, then i think we can rule the scheduler out.
> From what I can tell on the code, it likely means your wifi is connected, which would immediately exit the while loop.
> 
> Does that answer your question?

Hi!
I'm going to send you my while code, I think this can help you to understand my issue:
 
```
while(WiFiMulti.run() != WL_CONNECTED) 
  {
    if (serial == 1)
    {
      Serial.print(".");
    }
    
    digitalWrite(azul, LOW);
    delay(40);
    digitalWrite(azul, HIGH);
    delay(40);

  }
```
 Serial is printing for me the "." but I don't see the blink of the led "azul" (I tested with the router off, so ESP can't be able to connect)

If I delete this lines (That are before the while):

 ```
 Scheduler.start(&task_one);
 Scheduler.start(&task_two);
 Scheduler.begin();
```
While works perfectly, but multitasking obviously not works in that case.

I appreciate your help!
Thanks!

Giampaolo.

GiampaMan, 2018-12-05:
> Hi,
> 
> Sorry, could you post the entire file so I can see what is going on. I am unclear where the `while` and `Scheduler.begin()` are. If they are both in your main `setup` function, then they scheduler can have no effect of the code in the while, as it only runs afterwards. That would be super weird if it is having an effect on the code before it.

Hi!

Yes, this is a really weird issue, while is completely ok if you remove this lines: 
```
  Scheduler.start(&task_one);
  Scheduler.start(&task_two);
  Scheduler.begin();
```

This is the complete void setup code:
```
void setup() {
  
  EEPROM.begin(512);
  Serial.begin(115200);
  
  WiFiMulti.addAP(MySSID1, MyWifiPassword1);
  //WiFiMulti.addAP("xxxxxxx", "xxxxxxx");
  //WiFiMulti.addAP(MySSID2, MyWifiPassword2);
  //WiFiMulti.addAP(MySSID3, MyWifiPassword3);

  if (serial == 1)
  {
    Serial.println();
    Serial.print("Connecting to Wifi: ");
    Serial.println(MySSID1);  
  }
  while(WiFiMulti.run() != WL_CONNECTED) 
  {
    if (serial == 1)
    {
      Serial.print(".");
    }
    digitalWrite(azul, LOW);
    delay(40);
    digitalWrite(azul, HIGH);
    delay(40);
  }
  if(WiFiMulti.run() == WL_CONNECTED) 
  {
    if (serial == 1)
    {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
    }
    digitalWrite(azul, LOW);
  }

  Scheduler.start(&task_one);
  Scheduler.start(&task_two);
  Scheduler.begin();
}
```

GiampaMan, 2018-12-10:
Hi!
I found problem, was on my code, sorry.
Thanks for your help!
Have a nice day!

closed 2018-12-10 by GiampaMan

nrwiersma, 2018-12-10:
Glad you found the issue. Sorry I couldnt be more help.

GiampaMan, 2018-12-10:
> Glad you found the issue. Sorry I couldnt be more help.

Thank you for all!
One question more, is normal that the ESP crash? I added 4 tasks.
Thanks!

nrwiersma, 2018-12-10:
You have to check your image size, as the scheduler can take up a fair amount to ram and rom space. Other then that, no.

GiampaMan, 2018-12-18:
> You have to check your image size, as the scheduler can take up a fair amount to ram and rom space. Other then that, no.

Hi!
Sorry about the late response. I was pending to answer you.
Do you think this problem can be solved with a Wemos D1 Pro? This one has 16MB

nrwiersma, 2018-12-18:
Sorry, really cannot say. This will require debugging on your part.

GiampaMan, 2018-12-18:
> Sorry, really cannot say. This will require debugging on your part.

Hi! 
Thank you so much.
Have a nice day!