nrwiersma/ESP8266Scheduler #14: Overloadin loop()

troneleck, 2019-01-13:
schedul_test:6:10: error: 'void PrintTask::loop()' cannot be overloaded
     void loop()  {
          ^
schedul_test:62:6: error: with 'void PrintTask::loop()'
 void loop() {}
      ^
exit status 1
'void PrintTask::loop()' cannot be overloaded

I try the example file, but got this error.

nrwiersma, 2019-01-14:
Hi,

Can you tell me your setup, IDE, C++, versions?

PropsDataCom, 2019-01-14:
I am having the same problem.  I have sketch 1.8.8,  I have the default install, nothing changed from defaults.  have  the board as generic esp8266 module,  Flash Size; 1 meg (no spiffs)  I have no problem compiling other programs.

troneleck, 2019-01-14:
IDE = 1.8.7 WIN and 1.8.8 LINUX

ESP core = 2.4.2 WIN / 2.5 B2 LINUX

nodemcu  V3

Error is from your example.

nrwiersma, 2019-01-14:
This issue seems to be specific to the Arduino IDE. I have tried the code on PlatformIO (Atom), which is what I usually use, and it works fine. I am not sure why the Arduino IDE would cause trouble, but I have heard of other C++ issues in it.

Can you guys give PlatformIO a try and see if that fixes your issue?

troneleck, 2019-01-14:
I realy dont know how to use the other ide....but  installed something called VSCode with platformIO...
It´s much more complicated then Arduino IDE.
But with try and error i was able to make it work.
The terminal now displays:

Print Loop Start
Free Heap: 34904 bytes
Print Loop End
Free Heap: 34400 bytes
Print Loop Start
Print Loop End
Print Loop Start
Free Heap: 34400 bytes
Print Loop End
Print Loop Start
Free Heap: 34400 bytes
Print Loop End
Free Heap: 34232 bytes
Print Loop Start
Print Loop End
Free Heap: 34232 bytes
Print Loop Start
Print Loop End
Print Loop Start
Free Heap: 34232 bytes
Print Loop End
Print Loop Start
Free Heap: 34232 bytes

nrwiersma, 2019-01-14:
Awesome. I think the issue with Arduino IDE is the way it compiles, but there is little I can do about that.

nrwiersma, 2019-01-15:
Can we consider the issue closed then?