nrwiersma/ESP8266Scheduler #15: Arduino compatibility

eried, 2019-09-27:
Can this run on AVR? I always use your library for my ESP projects, but I am trying to make a small project with a screen that runs on both platforms. Any ideas?

nrwiersma, 2019-09-27:
I mean you could try, but i highly doubt it. It is based around the ESP8266 [cont.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/cont.h) to do the stack context switching. This is in turn based around the ESP8266 chip design i believe. Not even sure it would work on an ESP32.

eried, 2019-09-27:
ah, it requires extra work. I need to think on a solution maybe with something like: https://github.com/mikaelpatel/Arduino-Scheduler mixed with your wrapper. Or maybe some thin layer with #ifdef for both architectures :/

closed 2019-09-27 by eried