nrwiersma/ESP8266Scheduler #16: Any online request causing delay on the main loop

PawelSzymanski89, 2020-01-04:
I have one task with simple request to REST api and if i'll start in main esp loop is waiting for response...

`

class ApiRequest : public Task {
  public:
    void loop() {
      Serial.println("ApiRequest");
      if (WiFi.status() == WL_CONNECTED) {
        client.get("/basementData");
        int statusCode = client.responseStatusCode();
        String resBody = client.responseBody();
        //        Serial.println("-------------------------");
        //        Serial.println(resBody);
        if (statusCode = 200) {
          JsonObject root = doc.as<JsonObject>();
          auto error = deserializeJson(doc, resBody);
          if (!error) {
            //            double stoveTemp = root["stoveTemperature"];
            stoveTemperature = (int)root["stoveTemperature"];
            waterTemperature = (int)root["waterTemperature"];
            timeMills = root["timeStamp"];
          }
        }
      }
    }
} api_task;`

nrwiersma, 2020-01-05:
What is the issue exactly?

dbuezas, 2021-02-03:
`client.get` will wait using the global Delay, this blocks the tasks.

 The only ways around this I can think of are:
* Implement the get request yourself so you can call the task-specific delay function instead [code](https://arduino-esp8266.readthedocs.io/en/2.5.0/esp8266wifi/client-examples.html#now-to-the-sketch)
* Expose Scheduler.current->delay globally and wrap the library include with #define and #undefine to change the calls to global delay and yield to Scheduler.current->delay. (hacky but sounds fund)

dbuezas, 2021-02-06:
### I found a way to make this work:
1. expose the yield function of the current Task (e.g SchedulerClass::yield())
2. Make a new class that inherits from HTTPSClient
3. 
```cpp
class HTTPSClientScheduled : public HTTPSClient {
  int available() override {
    SchedulerClass::yield();
    return HTTPSClient::available();
  }
};
```
and instantiate a client from `HTTPSClientScheduled` instead.

This works because `GET` makes extensive use of `readStringUntil`, which in term calls `available`, which just happens to be overridable. I'm using it with [HTTPSRedirect](https://github.com/jbuszkie/HTTPSRedirect)

---

@nrwiersma would you consider a PR that exposes yield and delay in that way inside the Scheduler class?

nrwiersma, 2021-02-07:
@dbuezas In light of #20 it seems unlikely.

dbuezas, 2021-02-07:
I think it can be done (I first have to understand the run groups thing), I'll make a separate PR after #20

xyzshantaram, 2024-12-17:
@dbuezas @nrwiersma So what's the verdict? Does the workaround in https://github.com/nrwiersma/ESP8266Scheduler/issues/16#issuecomment-774521466 still work? There's a new LeanTask now right? I have the same issue - I want to do MIDI playback in one task, delay() until the next note and then go back to polling. If it does work then this issue should probably be closed

dbuezas, 2024-12-18:
I haven't used this in a long time, but the workaround should still work and be necessary

dbuezas, 2024-12-18:
Most likely WiFiClientSecure.