nrwiersma/ESP8266Scheduler #28: Scheduler incompatible with ESP8266-core-3.0.0

diedthreetimes, 2021-06-10:
Running the example code on a ESP-12e(NodeMCU) with the latest firmware produces the fails with following error:

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x402012af: SchedulerClass::start(Task*) at .../libraries/ESP8266Scheduler/src/Scheduler.cpp line 23
EXCVADDR: 0x00000004

Decoding stack results
0x402011c5: setup() at ~/Documents/Arduino/compatibility_test/compatibility_test.ino line 55
0x40201cc4: loop_wrapper() at ~/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/core_esp8266_main.cpp line 19
```

This worked fine last month, so I'm assuming the issue from upgrading [the core](https://github.com/esp8266/Arduino) to [3.0.0](https://github.com/esp8266/Arduino/releases/tag/3.0.0). Any ideas?

nrwiersma, 2021-06-10:
Hi,

Looking at other issues like this on github, it seems possible that you ran out of Heap memory. Perhaps the new core uses more Heap than before?

diedthreetimes, 2021-06-11:
I don't think so. I just ran the example code listed in the README.md. While error 29 is usually a memory issue I think that in this case it's something specific to the internals of the scheduler.

FWIW, my specific code works with another scheduler that doesn't rely on "cont.h" without issues. 

Looking at the commit for v0.1.0 my guess is that this has something to do with the `static Task main` although I have no idea why this memory would not be writable.

diedthreetimes, 2021-06-11:
I did a bit more digging. Notice in the crash report: `EXCVADDR=0x00000004`. According to [this memory map](https://www.esp8266.com/wiki/doku.php?id=esp8266_memory_map) that is indeed non-writable memory. 

[This article](https://esp32.com/viewtopic.php?t=12341) suggests that it's likely indicative of writing to a member variable of a NULL ptr. If this really is the case (and it seems to be) then it likely is an instance of the static initialization order fiasco and I'd recommend refactoring away all of the static variables (you don't really need any for the singleton method to work). 

Here is likely the order of events which caused this crash (and which changes based on the number and size of includes). 

1. `SchedulerClass Scheduler;` initialized calling the constructor and executing `main.prev = &main;` since main is not yet initialized and `&main == NULL`.
2. `Task SchedulerClass::main;` initialized, meaning `&main` exists and is not null.

diedthreetimes, 2021-06-11:
Then again there's something very wrong with this logic because if it were true I would expect step 1 to fail. I expect there is some kind of magic preventing this in the compiler. Despite this I am fairly confident limiting the use of statics will prevent this issue. 

This initialization order issue is the main reason complex statics are forbidden in [Google's C++ style guide](https://google.github.io/styleguide/cppguide.html#Static_and_Global_Variables).

nrwiersma, 2021-06-11:
In the `master` branch there is a version that gets rid of the `main` task, can you give this a try? If it solves the issue I can release it ASAP (which I have been meaning to do anyway).

diedthreetimes, 2021-06-11:
You were right, this works on master.

closed 2021-06-11 by diedthreetimes

nrwiersma, 2021-06-11:
I will release master in a couple of hours.