nrwiersma/ESP8266Scheduler #32: non-blocking delay in non-member function

jeremias-jordan, 2022-05-28:
Hi there,

thanks for the library. I wanted to ask if there is any chance to use delays of called non-member functions as non-blocking delays for the scheduler. I just can't get it to work. 

I could provide a full-code-example but for the sake of simplicity, this would be a very basic example of what i am trying to do. 

```

class SomeTask : public Task {
protected:
    void loop() {
        nonmemberfunction();
    }
} some_task;

void nonmemberfunction(){
   do_something();
   delay(500); // i want to use this delay for other processes
}

```

nrwiersma, 2022-05-29:
Hi,

There is a trick in the `Task` that exposes a delay function that overrides the "normal" delay as a protected method. To use it outside the `Task` you would need to pass it through to the function, or expose a method on your task that will call the protected delay and pass your `Task`.

jeremias-jordan, 2022-05-30:
Thank you very much! I exposed a function on my Task and it worked :)
Do you have any idea how to also achieve this with external libraries that make use of the delay function?

nrwiersma, 2022-05-30:
As far as I can tell, it would not be possible.

jeremias-jordan, 2022-06-01:
No problem, thank you anyway for the library and your help!

closed 2022-06-01 by jeremias-jordan