nrwiersma/ESP8266Scheduler #38: cont_yield with ESP8266 Release 3.1.0

Chris685, 2023-01-08:
Software is no longer buildable with new Release of ESP8266 (https://github.com/esp8266/Arduino).

```
In file included from Scheduler.cpp:2:
Task.h: In member function 'virtual void Task::yield()':
Task.h:20:5: error: 'cont_yield' was not declared in this scope; did you mean 'can_yield'?
   20 |     cont_yield(&context);
      |     ^~~~~~~~~~
      |     can_yield

nrwiersma, 2023-01-08:
`cont_yeild` was renamed to `cont_suspend` in this commit: https://github.com/esp8266/Arduino/commit/c312a2eaf1356ceaafad7c4935fa850e087c84fe#diff-1e7aa14c3546219ddea33c784ef21dfcfe958d0a7c9fba700cadec77c09a8123L63

Chris685, 2023-01-08:
Ok, sounds easy.

Will you also perform an Update or should i make an PullRequest?

File Task.h
```
  void yield() override {
    cont_suspend(&context);
  }

nrwiersma, 2023-01-08:
A pull request would be welcome.

nrwiersma, 2023-01-09:
Fixed by #40

closed 2023-01-09 by nrwiersma