nrwiersma/ESP8266Scheduler #44: cont_suspend was not declared in this scope

RJD02, 2023-03-06:
Hey,
I am new to the scheduling stuff, but I tried out your simple example and I'm getting this error(screenshot attached)
[image]

The code is the same as simple example posted on main github page of this repo.
Possible Cause: As far as I researched, it seems to be the cont.h header file problem, since I checked out the cont.h file from Arduino and turns out, there is cont_suspend method in it. But I guess, my IDE is not finding the reference to it.

Please help me find the issue

Thanks!

nrwiersma, 2023-03-06:
Hi.

I dont know how your system is setup or how you are compiling, so I am not able to debug this.

RJD02, 2023-03-06:
Fixed it
Process: Found out that inside cont.h header file, there's no cont_suspend method(at least for me), but there is a cont_yield method. I changed the cont_suspend to cont_yield inside the Task.h header file. And now it works fine.

Thanks

nrwiersma, 2023-03-06:
This means you are running an old version of the Arduino framework. The lib currently supports 3.1.0 and above. You can use an earlier version for support for the older framework.

closed 2023-03-07 by nrwiersma