nrwiersma/ESP8266Scheduler #46: Build Issue with PlatformIO

camdry, 2024-01-07:
Hi there!
Thanks for providing this code, I can't wait to see it working!

I'm having an issue when building my project including this library using the PlatformIO Visual Studio Code extension.
I have not edited any code in the header files so they appear as in the master branch but there are 7 build issues in Task.h when building my project.

Here is my platformio.ini file

```
[platformio]
default_envs = nodemcuv2

[common]
build_flags = 
	-D VERSION=1.2.3
	-D DEBUG=1

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = 
	${common.build_flags}
	-I$PROJECT_DIR/include
	-fexceptions
lib_deps = 
	fastled/FastLED@^3.6.0
	knolleary/PubSubClient@^2.8
	arduino-libraries/Arduino_JSON@^0.2.0
	nrwiersma/ESP8266Scheduler @ ^1.0
monitor_speed = 115200
test_ignore = test_desktop
```
And here are the build issues reported in Task.h
[image]

I've tried deleting and re-adding the library and also restarting my machine but no joy. Is there anything you could suggest?
Please let me know if you need any further information from me :)

Thanks!
Cam

nrwiersma, 2024-01-08:
Hi,

It is possible it is an issue with your C version. It seems to not like the inheritance from `AbstractTask`.

camdry, 2024-02-11:
Hi there

Apologies for my delayed response, thank you for getting back to me on this!
I have just updated my version of C++ but this has not resolved the issue after restarting my machine and reinstalling the library in to my project.

Here is the latest version that I installed:
[image]

Is this what you were referring to? :)

Thanks!
Cam

camdry, 2024-02-11:
Hi again

I think I have been able to resolve this by reordering the #include statements in one of my header files.
[image]

I'm not sure if this is a good workaround or not as my build is still failing with linking errors from the firmware elf.

Please see below the output from my currently failing build. I don't think these have anything to do with your library:

I'm happy to mark this as closed if nothing further comes to mind :) 

`Linking .pio\build\nodemcuv2\firmware.elf`
`c:/users/camer/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o:(.text.setup+0x18): undefined reference to _ZN13FastLedHelper14initializeLEDsEv'`
`c:/users/camer/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\main.cpp.o: in function setup':main.cpp:(.text.setup+0x5a): undefined reference to _ZN10MqttHelper9setupMQTTER12PubSubClient'`
`c:/users/camer/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\tasks\fastledtask.cpp.o:(.text._ZN11FastLedTask4loopEv+0x4): undefined reference to _ZN19ColourToStatMapping13AverageStressE'`
`c:/users/camer/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\tasks\fastledtask.cpp.o: in function _ZN11FastLedTask4loopEv':fastledtask.cpp:(.text._ZN11FastLedTask4loopEv+0xd2): undefined reference to _ZN13FastLedHelper12displayStatsE6Stringi4CRGBmR11StateObject'`
`c:/users/camer/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: fastledtask.cpp:(.text._ZN11FastLedTask4loopEv+0x11a): undefined reference to _ZN13FastLedHelper12displayStatsE6Stringi4CRGBmR11StateObject'`
`c:/users/camer/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\src\tasks\fastledtask.cpp.o:fastledtask.cpp:(.text._ZN11FastLedTask4loopEv+0x1c2): more undefined references to _ZN13FastLedHelper12displayStatsE6Stringi4CRGBmR11StateObject' follow`
`collect2.exe: error: ld returned 1 exit status`
`*** [.pio\build\nodemcuv2\firmware.elf] Error 1`

closed 2024-02-11 by camdry